ATrappmann/PN5180-Library #3: Wiring, compiling, and LEDs!

alastaira, 2018-10-22:
Hi!

I invested in a PN5180 board a few months back, fiddled with it for a bit but didn't get very far and then other things came along and it got put back in the drawer. Just pulled it out again and started to tinker and was delighted to find you'd started this project in the meantime - I'd like to contribute in some way if I can help. 
To start, I attach a wiring diagram for Arduino UNO drawn up in Fritzing - thought you might find it helpful to put on the homepage. 
Also, I note that the current PN5180-library.ino sketch does not compile using the Arduino IDE (1.8.5, under Windows 10 64bit)) - the `goto` statements cause several "crosses initialization" errors. I've managed a quick-workaround adding a boolean "errorHasOccurred" flag to the top of the loop() and replacing all the `goto error` references with: 
```
if(errorHasOccurred) { 
  showIRQStatus();
  nfc.reset();
  nfc.setupRF();
  return;
}
```
I'll probably work up a more elegant solution, but that allows the code to compile (and getting rid of the `goto`s is probably no bad thing anyway). I then get this in the serial monitor:

```
PN5180 Hard-Reset...
----------------------------------
Reading product version...
Product version=3.5
----------------------------------
Reading firmware version...
Firmware version=3.5
----------------------------------
Reading EEPROM version...
EEPROM version=145.0
----------------------------------
Reading IRQ pin config...
IRQ_PIN_CONFIG=0x1
----------------------------------
Reading IRQ_ENABLE register...
IRQ_ENABLE=0x1
----------------------------------
Enable RF field...
```
but from then on fails to read any tags. Also I note there are no LEDs lit on my PN5180 board, and I swear there used to be! Can you confirm if/when the SMD LEDS labelled L1, L2, L3 should be lit? Thanks.
[image]

alastaira, 2018-10-22:
Thanks for the quick response! Ah yes, you might be right about L1-L3 (my eyesight is not great, so surface-mounted components all look much the same to me!) - I could've sworn that I used to have an LED on the board, but I've been fiddling with lots of different RFID readers, so probably just got confused.

That schematic and code snippet are really useful, thanks - I'll carry on tinkering :)

closed 2018-10-22 by alastaira

offdbster, 2019-03-12:
Hi .. i try to use this library for reading ISO15693 tag but the only response is 

Error in getInventory: No card detected!
IRQ-Status 0x6: [ TX IDLE ]
*** No card detected!

any suggestion ??

ATrappmann, 2019-03-12:
Please enable debugging in the library. 

In file PN5180ISO15693.cpp:
- uncomment:  #define DEBUG 1

If that still gives no hints, do the same in file PN5180.cpp

—Andreas

offdbster, 2019-03-12:
Hi Andrea 
Thank you for the answer
I have enabled the DEBUG in PN5180.cpp file and the output is:

==================================
Uploaded: Mar 12 2019 18:43:43
PN5180 ISO15693 Demo Sketch
SPI pinout: SS=5, MOSI=23, MISO=19, SCK=18
----------------------------------
PN5180 Hard-Reset...
Read IRQ-Status register...
Reading register 0x02...
Sending SPI frame: '04 02'
Receiving SPI frame...
Received: 04 00 00 00'
Register value=0x00000004
IRQ-Status=0x00000004
Clear IRQ-Status with mask=xFFFFFFFF
Write Register 0x03, value (LSB first)=0xFFFFFFFF
Sending SPI frame: '00 03 FF FF FF FF'
----------------------------------
Reading product version...
Reading EEPROM at 0x10, size=2...
Sending SPI frame: '07 10 02'
Receiving SPI frame...
Received: 05 03'
EEPROM values: 05 03 
Product version=3.5
----------------------------------
Reading firmware version...
Reading EEPROM at 0x12, size=2...
Sending SPI frame: '07 12 02'
Receiving SPI frame...
Received: 05 03'
EEPROM values: 05 03 
Firmware version=3.5
----------------------------------
Reading EEPROM version...
Reading EEPROM at 0x14, size=2...
Sending SPI frame: '07 14 02'
Receiving SPI frame...
Received: 00 91'
EEPROM values: 00 91 
EEPROM version=145.0
----------------------------------
Enable RF field...
Load RF-Config: txConf=0D, rxConf=8D
Sending SPI frame: '11 0D 8D'
Set RF ON
Sending SPI frame: '16 00'
Read IRQ-Status register...
Reading register 0x02...
Sending SPI frame: '04 02'
Receiving SPI frame...
Received: 00 02 00 00'
Register value=0x00000200
IRQ-Status=0x00000200
Clear IRQ-Status with mask=x00000200
Write Register 0x03, value (LSB first)=0x00020000
Sending SPI frame: '00 03 00 02 00 00'
Write Register 0x00 with AND mask (LSB first)=0xF8FFFFFF
Sending SPI frame: '02 00 F8 FF FF FF'
Write Register 0x00 with OR mask (LSB first)=0x03000000
Sending SPI frame: '01 00 03 00 00 00'
----------------------------------
Loop #0
Send data (len=3): 26 01 00
Write Register 0x00 with AND mask (LSB first)=0xF8FFFFFF
Sending SPI frame: '02 00 F8 FF FF FF'
Write Register 0x00 with OR mask (LSB first)=0x03000000
Sending SPI frame: '01 00 03 00 00 00'
Get Transceive state...
Reading register 0x1D...
Sending SPI frame: '04 1D'
Receiving SPI frame...
Received: 00 00 02 01'
Register value=0x01020000
TRANSCEIVE_STATE=0x01
Sending SPI frame: '09 00 26 01 00'
Read IRQ-Status register...
Reading register 0x02...
Sending SPI frame: '04 02'
Receiving SPI frame...
Received: 06 00 00 00'
Register value=0x00000006
IRQ-Status=0x00000006
Error in getInventory: No card detected!
Read IRQ-Status register...
Reading register 0x02...
Sending SPI frame: '04 02'
Receiving SPI frame...
Received: 06 00 00 00'
Register value=0x00000006
IRQ-Status=0x00000006
IRQ-Status 0x6: [ TX IDLE ]
*** No card detected!
Read IRQ-Status register...
Reading register 0x02...
Sending SPI frame: '04 02'
Receiving SPI frame...
Received: 04 00 00 00'
Register value=0x00000004
IRQ-Status=0x00000004
Clear IRQ-Status with mask=xFFFFFFFF
Write Register 0x03, value (LSB first)=0xFFFFFFFF
Sending SPI frame: '00 03 FF FF FF FF'
Load RF-Config: txConf=0D, rxConf=8D
Sending SPI frame: '11 0D 8D'
Set RF ON
Sending SPI frame: '16 00'
Read IRQ-Status register...
Reading register 0x02...
Sending SPI frame: '04 02'
Receiving SPI frame...
Received: 00 02 00 00'
Register value=0x00000200
IRQ-Status=0x00000200
Clear IRQ-Status with mask=x00000200
Write Register 0x03, value (LSB first)=0x00020000
Sending SPI frame: '00 03 00 02 00 00'
Write Register 0x00 with AND mask (LSB first)=0xF8FFFFFF
Sending SPI frame: '02 00 F8 FF FF FF'
Write Register 0x00 with OR mask (LSB first)=0x03000000
Sending SPI frame: '01 00 03 00 00 00'
----------------------------------
Loop #1
Send data (len=3): 26 01 00
Write Register 0x00 with AND mask (LSB first)=0xF8FFFFFF
Sending SPI frame: '02 00 F8 FF FF FF'
Write Register 0x00 with OR mask (LSB first)=0x03000000
Sending SPI frame: '01 00 03 00 00 00'
Get Transceive state...
Reading register 0x1D...
Sending SPI frame: '04 1D'
Receiving SPI frame...
Received: 00 00 02 01'
Register value=0x01020000
TRANSCEIVE_STATE=0x01
Sending SPI frame: '09 00 26 01 00'
Read IRQ-Status register...
Reading register 0x02...
Sending SPI frame: '04 02'
Receiving SPI frame...
Received: 06 00 00 00'
Register value=0x00000006
IRQ-Status=0x00000006
Error in getInventory: No card detected!

offdbster, 2019-03-14:
This library is tested for ESP32 ?

ATrappmann, 2019-03-14:
Not from my side. There was a guy who used it that way.
I only use it on an regular Arduino Uno or Mega.

 
 
 

repository","main_image_url":"https://github.githubassets.com/images/email/message_cards/header.png","avatar_image_url":"https://github.githubassets.com/images/email/message_cards/avatar.png","action":{"name":"Open 
in 
GitHub","url":"https://github.com/ATrappmann/PN5180-Library"}},"updates":{"snippets":[{"icon":"PERSON","message":"@offdbster 
in #3: This library is tested for ESP32 ?\n\nIl mar 12 mar 2019, 17:00 
Andreas Trappmann \u003cnotifications@github.com\u003e 
ha\nscritto:\n\n\u003e Please enable debugging in the 
library.\n\u003e\n\u003e In file PN5180ISO15693.cpp:\n\u003e - uncomment: 
#define DEBUG 1\n\u003e\n\u003e If that still gives no hints, do the same 
in file PN5180.cpp\n\u003e\n\u003e —Andreas\n\u003e\n\u003e \u003e Am 
\u003cnotifications@github.com\u003e:\n\u003e \u003e\n\u003e \u003e Hi .. i 
try to use this library for reading ISO15693 tag but the only\n\u003e 
response is\n\u003e \u003e\n\u003e \u003e Error in getInventory: No card 
detected!\n\u003e \u003e IRQ-Status 0x6: [ TX IDLE ]\n\u003e \u003e *** No 
card detected!\n\u003e \u003e\n\u003e \u003e any suggestion ??\n\u003e 
GitHub\n\u003e 
\u003chttps://github.com/ATrappmann/PN5180-Library/issues/3#issuecomment-472057883\u003e,\n\u003e 
.\n\u003e\n"}],"action":{"name":"View 
Issue","url":"https://github.com/ATrappmann/PN5180-Library/issues/3#issuecomment-472757533"}}} 
"https://github.com/ATrappmann/PN5180-Library/issues/3#issuecomment-472757533", 
"url": 
"https://github.com/ATrappmann/PN5180-Library/issues/3#issuecomment-472757533", 
"name": "View Issue" }, "description": "View this Issue on GitHub", 
"publisher": { "@type": "Organization", "name": "GitHub", "url": 
"https://github.com" } } ]

commit 921d34f9 (2018-10-23): (Re-)make compile for Arduino IDE, fixes #3 (#4)

* make compile for ESP-32

* Update README.md

* ESP-32 compatibility: MSBFIRST is already defined in Arduino.h

* sendData: Allow to set number of bits of last byte to transmit

* Merge with master

* Add schematic for these chinese boards

* make compile on Arduino IDE
sendData: Allow to set the number of bits for last byte. ISO-14443 WAK/WUP commands have 7 bits