ATrappmann/PN5180-Library #6: Support for ISO 14443A/B

stormwin, 2019-01-13:
Do this device and this plugin have support for ISO 14443A/B NFC Standard?
I have this device and a tons of Mifare Classic/Ultralight cards and Ntags. How can i use this plugin to read from them?

Thanks

ATrappmann, 2019-01-14:
Hi,
thank you for your interest in the library. The device does have support 
for ISO14443 but I did not implement it in my library because I only had 
the demand for ISO15693 tags. Since there was no library available, I 
posted my implementation to Github. Feel free to add your code for ISO 
14443 support and we can integrate it. The best way would to make another 
subclass of the PN5180 class.
--Andreas.
 
 
 
-----Original-Nachricht-----
Betreff: [ATrappmann/PN5180-Library] Support for ISO 14443A/B (#6)
Datum: 2019-01-13T19:52:22+0100
Von: "Vlado Velichkovski" <notifications@github.com>
An: "ATrappmann/PN5180-Library" <PN5180-Library@noreply.github.com>
 
 
 
Do this device and this plugin have support for ISO 14443A/B NFC Standard?
I have this device and a tons of Mifare Classic/Ultralight cards and Ntags. 
How can i use this plugin to read from them?
Thanks
—
﻿

ATrappmann, 2019-01-14:
> Thanks for reply.
> As i can see from PN5180 docs, there is some docs how to configure setupRF
> function to work for ISO 14443.
> Also found in docs about Mifare Authentication process and how to
> implement, but i do not found a way how to read and write to Mifare and
> ntags. That is my biggest problem now.
> Definitely i will fork this project today and start to work on
> implementation of ISO 14443, as i need it for my project.

Hi Vlado,
 
try google "iso14443 pdf". Part 3 & 4 of the ISO should be the documents 
you are looking for.
 
2nd guess: try google "arduino iso14443 library" and see what other have 
implemented on other boards. The ISO protocol is the same. You could use 
that with the basic transmission functions from the PN5180 class.
 
Cheers,
--Andreas.

tueddy, 2019-12-03:
Feel free to test my branch: https://github.com/tueddy/PN5180-Library/tree/ISO14443

tueddy, 2020-02-08:
Found this document "Using the PN5180 without (NXP) library":
[AN12650.pdf](https://github.com/ATrappmann/PN5180-Library/files/4175126/AN12650.pdf)

It explains the SPI command sequence both for detecting ISO14443 & ISO15693 cards in detail...

ATrappmann, 2020-02-09:
Great!! Thank you.
—Andreas

freemont-mike, 2021-02-26:
https://github.com/tueddy/PN5180-Library used this library, works perfect for the mifare classic cards.
Here's my sketch:
```
**`
#include <PN5180.h>
#include <PN5180ISO14443.h>
#define PN5180_NSS  10
#define PN5180_BUSY 9
#define PN5180_RST  7
PN5180ISO14443 nfc14443(PN5180_NSS, PN5180_BUSY, PN5180_RST);

void setup() {
  Serial.begin(115200);
  nfc14443.begin();
  nfc14443.reset();
  uint8_t productVersion[2];
  nfc14443.readEEprom(PRODUCT_VERSION, productVersion, sizeof(productVersion));
  if (0xff == productVersion[1]) { // if product version 255, the initialization failed
    Serial.println(F("Initialization failed!?"));
  }
  uint8_t firmwareVersion[2];
  nfc14443.readEEprom(FIRMWARE_VERSION, firmwareVersion, sizeof(firmwareVersion));
  uint8_t eepromVersion[2];
  nfc14443.readEEprom(EEPROM_VERSION, eepromVersion, sizeof(eepromVersion));
  nfc14443.setupRF();
}

void loop() {
  Serial.print(F("Loop #"));
  #if defined(ARDUINO_ARCH_ESP32)  
    Serial.println("Free heap: " + String(ESP.getFreeHeap())); 
  #endif
  uint8_t uid[10];
  // check for ISO-14443 card
  nfc14443.reset();
  nfc14443.setupRF();
  if (nfc14443.isCardPresent()) {
    uint8_t uidLength = nfc14443.readCardSerial(uid);
    if (uidLength > 0) {
      Serial.print(F("ISO14443 card found, UID="));
      for (int i=0; i<uidLength; i++) {
        Serial.print(uid[i] < 0x10 ? " 0" : " ");
        Serial.print(uid[i], HEX);
      }
      Serial.println(F("----------------------------------"));
      delay(1000); 
      return;
    }
  } 
  Serial.println(F("*** No card detected!"));
}
`**
```

KoenPaas, 2021-03-31:
@tueddy any idea how to implement APDU commands?

ATrappmann, 2021-08-19:
Fixed.

closed 2021-08-19 by ATrappmann