ATrappmann/PN5180-Library #9: off-by-one error in readSingleBlock?

floe, 2019-03-04:
Thank you for this excellent library! I've been testing this with a random ISO15963 card, and I found that I would always get the following output when reading the card:
```
Inventory successful, UID=E0:4:1502A49B3F9
----------------------------------
System Info retrieved: blockSize=4, numBlocks=28
----------------------------------
Read block #0: 00 00 00 2A  ...*
Read block #1: 00 00 00 2A  ...*
Read block #2: 00 00 00 2A  ...*
Read block #3: 00 00 00 2A  ...*
Read block #4: 00 00 00 2A  ...*
Read block #5: 00 00 00 2A  ...*
Read block #6: 00 00 00 2A  ...*
Read block #7: 00 00 00 2A  ...*
Read block #8: 00 00 00 2A  ...*
Read block #9: 00 00 00 2A  ...*
Read block #10: 00 00 00 2A  ...*
Read block #11: 00 00 00 2A  ...*
Read block #12: 00 00 00 2A  ...*
Read block #13: 00 00 00 2A  ...*
Read block #14: 00 00 00 2A  ...*
Read block #15: 00 00 00 2A  ...*
Read block #16: 00 00 00 2A  ...*
Read block #17: 00 00 00 2A  ...*
Read block #18: 00 00 00 2A  ...*
Read block #19: 00 00 00 2A  ...*
Read block #20: 00 00 00 2A  ...*
Read block #21: 00 00 00 2A  ...*
Read block #22: 00 00 00 2A  ...*
Read block #23: 00 00 00 2A  ...*
Read block #24: 00 00 00 2A  ...*
Read block #25: 00 00 00 2A  ...*
Read block #26: 00 00 00 2A  ...*
Read block #27: 00 00 00 2A  ...*
```

Note the 0x2A - when I read the card with NFC TagInfo on a smartphone, all blocks show as 0x00000000. And it's also noteworthy that byte 5 of the tag address is also 0x2A. After a bit of digging and reading the ISO15693 spec, I think that this line (https://github.com/ATrappmann/PN5180-Library/blob/master/PN5180ISO15693.cpp#L131) should be `blockData[i] = resultPtr[i+1];` (instead of 2+i)?

ATrappmann, 2019-03-07:
Thank you for your feedback and sorry for the late answer. I had no time to reproduce the issue yet. 
Looks strange for sure. I am wondering if that has to do with the options bit I removed with the last issue. When I started the library I always had this custom options bit set and everything worked fine with my cards. Latley there was a note, that this bit produced write errors to another card which didn't support this. Therefor I cleared the bit for writing and reading...  Interesting.
I am glad you found a working solution and reported it back. I will fix it in the next days.

geoffrey-vl, 2019-05-20:
I can confirm that it has to do with enabling the option flags bit.
The supplied code fix is also working as intended.

closed 2019-11-13 by ATrappmann