ATrappmann/PN5180-Library #10: Arduino Zero Errors

RatGit, 2019-04-09:
Hi,

I tried to run this library with an Arduino Zero and [this PN5180 Board](https://github.com/ATrappmann/PN5180-Library/files/2502996/PN5180.R1.1-170710_SCH.PDF), but it generates the following error message:

> _Error in getInventory: No card detected!
> IRQ-Status 0x6: [ TX IDLE ]
> *** No card detected!_

I turned on DEBUG in the "PN5180ISO15693.cpp" file as recommended in one of the other issue threads, but the code would not compile?

> _/libraries/PN5180/1.1.0/PN5180ISO15693.h:60:30: error:   initializing argument 1 of 'const __FlashStringHelper* PN5180ISO15693::strerror(ISO15693ErrorCode)' [-fpermissive]
>    const __FlashStringHelper *strerror(ISO15693ErrorCode errno);_

What does "No Card Detected" actually mean?

ps. I have ensured that the pins are voltage compatible and that all the other SPI peripherals are deselected. I also set the RST line to HIGH in "setup()" so that the "nfc.reset()" function works properly:

_`digitalWrite(PN5180_RST, HIGH); // Not Reset, (should modify "nfc::begin()" function to do this instead)`_

Any help would be greatly appreciated...

ATrappmann, 2019-04-09:
Hi,
 
thank you for your feedback. I will try to help. I have no Arduino Zero, so 
I can only give you some hints.
1) The message "No card detected!" means what it says. The Reader send out 
an NFC-field and no Interrupt was generated to signal, that there is an 
ISO15693 card in the field.
I do not know your card reader, but you could check the datasheet, if it 
generates the IRQ RX_SOF_DET_IRQ_STAT. 
 
2) The compile error has to do with your 32bit board. It looks as if 
headers are different. For an Uno, __FlashStringHelper is declared in 
WString.h as follows:
 
class __FlashStringHelper;
#define F(string_literal) (reinterpret_cast<const __FlashStringHelper 
*>(PSTR(string_literal)))
 
3) Thanks for the feedback with the RST-line. I will add this to my code.
 
Hope this helps a bit...
--Andreas.

 
 
 

<https://github.com/ATrappmann/PN5180-Library/issues/10> , or mute the 
thread
repository","main_image_url":"https://github.githubassets.com/images/email/message_cards/header.png","avatar_image_url":"https://github.githubassets.com/images/email/message_cards/avatar.png","action":{"name":"Open 
in 
GitHub","url":"https://github.com/ATrappmann/PN5180-Library"}},"updates":{"snippets":[{"icon":"DESCRIPTION","message":"Arduino 
Zero Errors (#10)"}],"action":{"name":"View 
Issue","url":"https://github.com/ATrappmann/PN5180-Library/issues/10"}}} [ 
"https://github.com/ATrappmann/PN5180-Library/issues/10", "url": 
"https://github.com/ATrappmann/PN5180-Library/issues/10", "name": "View 
Issue" }, "description": "View this Issue on GitHub", "publisher": { 
"@type": "Organization", "name": "GitHub", "url": "https://github.com" } } 
]

RatGit, 2019-04-10:
Hi Andreas,

Thank you for responding to my initial issue submission. I have subsequently decided to proceed with an Arduino Uno instead for my testing.
I duplicated your original wiring connections, (using the Uno with an identical PN5180 board and a 3.3V-5V level shifter) and used the Arduino IDE to compile and upload your library's example sketch.

The software still isn't able to communicate properly with the PN5180 board and eventually "hangs"...

```
==================================
Uploaded: Apr 10 2019 17:12:31
PN5180 ISO15693 Demo Sketch
----------------------------------
PN5180 Hard-Reset...
----------------------------------
Reading product version...
Product version=3.7
----------------------------------
Reading firmware version...
Firmware version=3.7
----------------------------------
Reading EEPROM version...
EEPROM version=51.0
----------------------------------
Enable RF field...
Loading RF-Configuration...
done.
Turning ON RF field...
done.
----------------------------------
Loop #0
Get Inventory...
Issue Command 0x01...
Error in getInventory: ISO15693ErrorCode=-1
No card detected!
IRQ-Status 0x30300: [ TX_RFOFF TX_RFON RX_SC_DET ]
*** No card detected!
Loading RF-Configuration...
done.
Turning ON RF field...
done.
----------------------------------
Loop #1
Get Inventory...
Issue Command 0x01...
Error in getInventory: ISO15693ErrorCode=-1
No card detected!
IRQ-Status 0x30100: [ TX_RFOFF RX_SC_DET ]
*** No card detected!
Loading RF-Configuration...
done.
Turning ON RF field...
done.
----------------------------------
Loop #2
Get Inventory...
Issue Command 0x01...
RX-Status=00181D7C, len=380
```

I'm not sure what exactly might be happening here and I am still working on it... I would certainly appreciate any insights you may be able to offer,

Kind Regards,
Keith.

ATrappmann, 2019-04-10:
Hi Keith,
 
did you see, that in Loop #2 the card was found? The snipped you send me 
ended with:

Loop #2 Get Inventory... 
Issue Command 0x01... 
RX-Status=00181D7C, len=380 

There should have been an positive answer from Get Inventory...!? The 
reader received 380 bytes of data!
 
Best Regards,
--Andreas.

 
 
 

GitHub","url":"https://github.com/ATrappmann/PN5180-Library"}},"updates":{"snippets":[{"icon":"PERSON","message":"@RatGit 
in #10: Hi Andreas,\r\n\r\nThank you for responding to my initial issue 
instead for my testing.\r\nI duplicated your original wiring connections, 
sketch.\r\n\r\nThe software still isn't able to communicate properly with 
\"hangs\"...\r\n\r\n```\r\n==================================\r\nUploaded: 
Apr 10 2019 17:12:31\r\nPN5180 ISO15693 Demo 
Sketch\r\n----------------------------------\r\nPN5180 
Hard-Reset...\r\n----------------------------------\r\nReading product 
version...\r\nProduct 
version=3.7\r\n----------------------------------\r\nReading firmware 
version...\r\nFirmware 
version=3.7\r\n----------------------------------\r\nReading EEPROM 
version...\r\nEEPROM 
version=51.0\r\n----------------------------------\r\nEnable RF 
field...\r\nLoading RF-Configuration...\r\ndone.\r\nTurning ON RF 
field...\r\ndone.\r\n----------------------------------\r\nLoop #0\r\nGet 
Inventory...\r\nIssue Command 0x01...\r\nError in getInventory: 
ISO15693ErrorCode=-1\r\nNo card detected!\r\nIRQ-Status 0x30300: [ TX_RFOFF 
TX_RFON RX_SC_DET ]\r\n*** No card detected!\r\nLoading 
RF-Configuration...\r\ndone.\r\nTurning ON RF 
field...\r\ndone.\r\n----------------------------------\r\nLoop #1\r\nGet 
Inventory...\r\nIssue Command 0x01...\r\nError in getInventory: 
ISO15693ErrorCode=-1\r\nNo card detected!\r\nIRQ-Status 0x30100: [ TX_RFOFF 
RX_SC_DET ]\r\n*** No card detected!\r\nLoading 
RF-Configuration...\r\ndone.\r\nTurning ON RF 
field...\r\ndone.\r\n----------------------------------\r\nLoop #2\r\nGet 
Inventory...\r\nIssue Command 0x01...\r\nRX-Status=00181D7C, 
len=380\r\n```\r\n\r\nI'm not sure what exactly might be happening here and 
may be able to offer,\r\n\r\nKind 
Regards,\r\nKeith.\r\n"}],"action":{"name":"View 
"https://github.com/ATrappmann/PN5180-Library/issues/10#issuecomment-481625761", 
"https://github.com/ATrappmann/PN5180-Library/issues/10#issuecomment-481625761", 
"https://github.com" } } ]

RatGit, 2019-04-11:
Hi Andreas,

Thank you for responding to my initial issue submission. I have subsequently decided to proceed with an Arduino Uno instead for my testing.
I duplicated your original wiring connections, (using the Uno with an identical PN5180 board and a 3.3V-5V level shifter) and used the Arduino IDE to compile and upload your library's example sketch.

The software still isn't able to communicate properly with the PN5180 board and eventually "hangs"...

```
==================================
Uploaded: Apr 10 2019 17:12:31
PN5180 ISO15693 Demo Sketch
----------------------------------
PN5180 Hard-Reset...
----------------------------------
Reading product version...
Product version=3.7
----------------------------------
Reading firmware version...
Firmware version=3.7
----------------------------------
Reading EEPROM version...
EEPROM version=51.0
----------------------------------
Enable RF field...
Loading RF-Configuration...
done.
Turning ON RF field...
done.
----------------------------------
Loop #0
Get Inventory...
Issue Command 0x01...
Error in getInventory: ISO15693ErrorCode=-1
No card detected!
IRQ-Status 0x30300: [ TX_RFOFF TX_RFON RX_SC_DET ]
*** No card detected!
Loading RF-Configuration...
done.
Turning ON RF field...
done.
----------------------------------
Loop #1
Get Inventory...
Issue Command 0x01...
Error in getInventory: ISO15693ErrorCode=-1
No card detected!
IRQ-Status 0x30100: [ TX_RFOFF RX_SC_DET ]
*** No card detected!
Loading RF-Configuration...
done.
Turning ON RF field...
done.
----------------------------------
Loop #2
Get Inventory...
Issue Command 0x01...
RX-Status=00181D7C, len=380
```

I'm not sure what exactly might be happening here and I am still working on it... I would certainly appreciate any insights you may be able to offer,

ps. I did notice that card was detected, however there was no card actually near the device?

Kind Regards,
Keith.

RatGit, 2019-04-11:
Hi Andreas,

I now have it all working on both the Uno and Zero devices... The level shifter boards I was using were broken. I am using the Sloeber Arduino Eclipse plugin as my development environment with an Atmel ICE, so I have hardware debugging working under Linux. (Note: You can also do this under Windows with Sloeber and Atmel Studio).

FYI: I am currently modifying the test code to support multiple devices hosted by the same Arduino.

Kind Regards,
Keith.

ATrappmann, 2019-04-11:
Hi Keith,
 
I am glad you found the problem and have a running solution now.
Enjoy.
 
--Andreas.

 
 
 

in #10: Hi Andreas,\r\n\r\nI now have it all working on both the Uno and 
can also do this under Windows with Sloeber and Atmel Studio).\r\n\r\nFYI: 
by the same Arduino.\r\n\r\nKind 
Regards,\r\nKeith."}],"action":{"name":"View 
"https://github.com/ATrappmann/PN5180-Library/issues/10#issuecomment-481969624", 
"https://github.com/ATrappmann/PN5180-Library/issues/10#issuecomment-481969624", 
"https://github.com" } } ]

RatGit, 2019-04-12:
Hi Andreas,

FYI: I've managed to modify the library to work with multiple PN5180 Reader/Writers. During this process I've modified your Library and Example sketch to support this, (see the attached Zip archive).
Note: I've converted your Arduino sketch file to a C++ module and associated header file.
The library changes include the following:

- Remove call to "**SPI.begin**" from the "**PN5180::begin**" function.
- Set the RST line to HIGH in the "**PN5180::setup**" function so that the "**PN5180::reset**" function works properly.

I've also found and fixed a number of minor coding errors... (see the list of changes at the end of this message):

Finally I have two outstanding issues.
 - The **first issue** is that in my call to setup the second PN5180 it always "hangs" in the final while loop of the "**PN5180::setRF_on**" function. For the moment I have dummied this out with a 100msec delay, (see code excerpt below). I'd appreciate it if you can suggest a better fix. 

```
bool PN5180::setRF_on() {
  PN5180DEBUG(F("Set RF ON\n"));

  uint8_t cmd[2] = { PN5180_RF_ON, 0x00 };

  SPI.beginTransaction(PN5180_SPI_SETTINGS);
  transceiveCommand(cmd, 2);
  SPI.endTransaction();

//  while (0 == (TX_RFON_IRQ_STAT & getIRQStatus())); // wait for RF field to set up
  delay(100);  // Replace while loop with a straight delay because the setup of the second PN5180 reader will hang on the while loop, (not sure why?)
  clearIRQStatus(TX_RFON_IRQ_STAT);
  return true;
}
```

 - The **second issue** I'm having is that I can't get a Write operation to work. If you have any code examples I would also appreciate it thanks. My attached code contains my failed attempt at doing this, (jn the "**writeNFC**" function). 

================================================================================
**List of Coding Fixes:**

**Line 249: PN5180.cpp**
_/root/Documents/sloeber-workspace/libraries/PN5180/1.1.0/PN5180.cpp:250:13: warning: comparison is always false due to limited range of data type [-Wtype-limits]
   if (len > 260) {
             ^_
`bool PN5180::sendData(uint8_t *data, size_t len, uint8_t validBits)`

**Line 432: PN5180ISO15693.cpp**
`ISO15693ErrorCode PN5180ISO15693::issueISO15693Command(uint8_t *cmd, size_t cmdLen, uint8_t **resultPtr)`

**Line 115: PN5180ISO15693.cpp**
_/root/Documents/sloeber-workspace/libraries/PN5180/1.1.0/PN5180ISO15693.cpp:115:41: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (int i=0; i<sizeof(readSingleBlock); i++) {
                                         ^_
`  for (int i=0; i<(int)sizeof(readSingleBlock); i++) {`

**Line 278: PN5180ISO15693.cpp**
_/root/Documents/sloeber-workspace/libraries/PN5180/1.1.0/PN5180ISO15693.cpp:278:33: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (int i=0; i<sizeof(sysInfo); i++) {
                                 ^_
`  for (int i=0; i<(int)sizeof(sysInfo); i++) {`

**Line 486: PN5180ISO15693.cpp**
_/root/Documents/sloeber-workspace/libraries/PN5180/1.1.0/PN5180ISO15693.h:60:30: error:   initializing argument 1 of 'const __FlashStringHelper* PN5180ISO15693::strerror(ISO15693ErrorCode)' [-fpermissive]
   const __FlashStringHelper *strerror(ISO15693ErrorCode errno);
                              ^_
`PN5180DEBUG(strerror((ISO15693ErrorCode)errorCode));`

RatGit, 2019-04-12:
**ps.** There were also some unused variable warnings...

_/root/Documents/sloeber-workspace/libraries/PN5180/1.1.0/PN5180ISO15693.cpp:308:13: warning: unused variable 'dsfid' [-Wunused-variable]
     uint8_t dsfid = *p++;
             ^_

_/root/Documents/sloeber-workspace/libraries/PN5180/1.1.0/PN5180ISO15693.cpp:351:14: warning: unused variable 'viccMemSize' [-Wunused-variable]
     uint16_t viccMemSize = (*blockSize) * (*numBlocks);
              ^_

_/root/Documents/sloeber-workspace/libraries/PN5180/1.1.0/PN5180ISO15693.cpp:366:13: warning: unused variable 'icRef' [-Wunused-variable]
     uint8_t icRef = *p++;
             ^_

Boilevin, 2019-05-26:
Hi.
I Don't know if it can help but i have this kind of trouble when wiring a ESP32 .
The wiring description in the code ommit the +5V so on ESP32Devkit powered on USB the +5V need to be wired with Vin .

Take a look at R13 and R15 on schematics.
Thanks.