ATrappmann/PN5180-Library #15: How to switch to next slot

key8931, 2019-07-05:
I work with ISO 15693, and I want to work with 16 slots and always no response, the doc says I need send EOF, but how?

ma-ze, 2019-07-24:
I'm also interested in this.
For me it would be enough to recognize multiple UIDs at the same time but sadly not getting that to work.

NicM1989, 2019-08-26:
> I work with ISO 15693, and I want to work with 16 slots and always no response, the doc says I need send EOF, but how?

Hi key8931. Can I ask what you've tried so far to implement the 16 slot read?
I am looking into adding to Trappmann's code to be able to handle multiple tag reads (anticollision). As far as I can make out the existing examples and libraries are setup to handle 1 card read at once and won't implement anticollision handling.... please correct me if i'm wrong.
The below is taken from the PN5180ISO15693.cpp library:
I'm assuming you've set bit 6 of the Request Flags byte to 1 and tried to perform some extra read functions to handle to returned data?
I'll keep looking into it and let you know if I have any luck. 
_____________________________________________________________________________________________________
* ISO 15693 - Protocol
 *
 * General Request Format:
 *  SOF, Req.Flags, Command code, Parameters, Data, CRC16, EOF
 *
 *  Request Flags:
 *    xxxx.3210
 *         |||\_ Subcarrier flag: 0=single sub-carrier, 1=two sub-carrier
 *         ||\__ Datarate flag: 0=low data rate, 1=high data rate
 *         |\___ Inventory flag: 0=no inventory, 1=inventory
 *         \____ Protocol extension flag: 0=no extension, 1=protocol format is extended
 *
 *  If Inventory flag is set:
 *    7654.xxxx
 *     ||\_ AFI flag: 0=no AFI field present, 1=AFI field is present
 *     |\__ Number of slots flag: 0=16 slots, 1=1 slot
 *     \___ Option flag: 0=default, 1=meaning is defined by command description

JelleME, 2019-12-03:
Hi everyone, I too am interested in an anticollision protocol since I need to detect multiple UIDs at the same time. However I have had no luck at using the 16 slots, which would be the proper way to implement anticollision. 

For now I have found a workaround by using the Quiet state of the ISO-15693 tags (I use ICODE-SLI tags but SLIX or SLIX2 should also have this option), see figure.

[image]

By setting the address flag and putting the tags to the Quiet state the tag will still respond to addressed commands as long as the inventory flag is not set. Keep in mind that this workaround will only work because in my application I know the tag UIDs beforehand, which allows me to put them to the Quiet state blindly (by sending the stay Quiet commands for all known UIDs). If you have a similar situation, I hope this method maybe useful to you.

I am however still interested in a proper anticollision protocol for anyone who makes progress on this!

Cheers.

tueddy, 2019-12-03:
An indian guy has a video for this and also an education software based on this library. It's like the NFC cockpit software and maybe helps to solve your problem? 
https://www.youtube.com/watch?v=0C7oAYbJQTA

JelleME, 2019-12-03:
Hi tueddy, thank you for the video, it did indeed give some good insights on more possibilities with this board. Unfortunately, as you said, the backend of his software is based on this library and in the video at 11:03 he specifically says using inventory slot 16 is not implemented. I may look into this a bit more when I have time.

NicM1989, 2019-12-04:
I also used the 'stay quiet' command as  a work-around like JelleME, but  yes this method won't work if you don' t know the UIDs of the multiple tags overlaying the reader. I managed to read five tags successfully (see images below) but haven't tested more than that.
 
[image]

[image]

alansbor, 2020-12-21:
Hi
An interesting cycle.
"Cycle sequence: 'select' 1st tag >> 'get inventory' 1st tag >> send 1st tag silently >> repeat for N tags >> start scan cycle again."

But how did you implement it, can you show your code?

grim-code, 2021-06-16:
Hi
In case someone finds it helpful, I got it thanks to the application note AN12650 of nxp in which they have an example in pseudocode.
[https://www.nxp.com.cn/docs/en/application-note/AN12650.pdf](url)