ATrappmann/PN5180-Library #17: Read ISO-15693 icode slix password protected tag

tueddy, 2019-10-21:
Hi Andreas,

thanks for your great library!
i want to read the UID of this password protected icode2 slix tag:

[image]

This tag is compatible to ISO-15693 but can be password protected. Once protected it does not response on any commands until it got the correct password.

To unlock the tag the workflow is:

1. Get Random command receives two bytes of random number
2.. XOR the known password with the previous received random number
3. Unlock the tag with SetPassword(XORred password from 2.) command
4. ReadInventory() and get UID as usual

For your library two commands missing:
1.. GET_RANDOM_NUMBER (command is 0xB2, 0x04)
2. SET_PASSWORD (command is 0xB3 0x04 0x04)

Any idea how to transfer the two commands?

This is what i tried, failure on first command 
GetRandom:

`    // get random number
    uint8_t getrandom[] = { 0xB2, 0x04};
    uint8_t *readBuffer;
    ISO15693ErrorCode rc = nfc.issueISO15693Command(getrandom, sizeof(getrandom), &readBuffer);
    if (ISO15693_EC_OK != rc) {
      Serial.println(F("*** GetRandom failed!"));
    }
`

Command fails. Any idea? Missing flags?
Thanks in advance
Dirk

tueddy, 2019-10-21:
Doc: [https://www.nxp.com/docs/en/data-sheet/SL2S2602.pdf](url)
Page 17/18

ATrappmann, 2019-10-22:
Hi Dirk,
 
I think you are at least missing the flags, which are required by the 
ISO15693 protocol.
Check the comment in PN5180ISO15693.cpp at lines 379ff. I think you sould 
try a command with:
 
{ 0x02, 0xB2, 0x04} // High data rate
{ 0x42, 0xB2, 0x04} // High data rate + Option Flag
 
 
What exact ISO15693ErrorCode is returend?  
 
Hope that helps.
--Andreas.

protected tag (#17)
Datum: 2019-10-21T23:13:08+0200
 
 
 

[IMG_7357]
<https://user-images.githubusercontent.com/11274319/67242948-de923780-f456-11e9-93c1-1d35be340564.jpg>
   2. Unlock the tag with SetPassword(XORred password from 2.) command
   3. ReadInventory() and get UID as usual
"https://github.com/ATrappmann/PN5180-Library/issues/17?email_source=notifications\u0026email_token=AEVOSHS5IYQSKMOGLQDFQI3QPYLNZA5CNFSM4JDFHCPKYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4HTKQM3Q", 
"url": 
"https://github.com/ATrappmann/PN5180-Library/issues/17?email_source=notifications\u0026email_token=AEVOSHS5IYQSKMOGLQDFQI3QPYLNZA5CNFSM4JDFHCPKYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4HTKQM3Q", 
"name": "View Issue" }, "description": "View this Issue on GitHub", 
"publisher": { "@type": "Organization", "name": "GitHub", "url": 
"https://github.com" } } ]

ATrappmann, 2019-10-22:
Hi Dirk,
 
are you sure that getInventory is already protected? From the spec it says:
 
"Password protected memory management (Read/Write access)"
 
From that I would say, only readSingleBlock and writeSingleBlock should be 
affected by the passoword.
Standard functions like getInventory should work without the password.
 
--Andreas.

protected tag (#17)
Datum: 2019-10-21T23:41:13+0200
 
 
 

"https://github.com/ATrappmann/PN5180-Library/issues/17?email_source=notifications\u0026email_token=AEVOSHTCMSXXDDOXLDTH4KLQPYOXJA5CNFSM4JDFHCPKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEB34J6Y#issuecomment-544720123", 
"https://github.com/ATrappmann/PN5180-Library/issues/17?email_source=notifications\u0026email_token=AEVOSHTCMSXXDDOXLDTH4KLQPYOXJA5CNFSM4JDFHCPKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEB34J6Y#issuecomment-544720123", 
"https://github.com" } } ]

tueddy, 2019-10-22:
Hi Andreas,

thanks for your fast response!
The tag does not respond to readInventory()
I will test again with the flags in the evening and report here.

Best
Dirk

tueddy, 2019-10-22:
Hi Andreas,

that helped a lot, wow, first time my little tiny tag responds:

` *** GetRandom failed!
*** GetRandom failed!
*** GetRandom failed!
GetRandom successful: 0x0:DC:12, CRC:0:0
GetRandom successful: 0x0:39:E6, CRC:0:0
GetRandom successful: 0x0:BE:19, CRC:0:0
GetRandom successful: 0x0:EB:F8, CRC:0:0
GetRandom successful: 0x0:D7:89, CRC:0:0
GetRandom successful: 0x0:92:5E, CRC:0:0
GetRandom successful: 0x0:3F:D8, CRC:0:0
GetRandom successful: 0x0:4D:36, CRC:0:0
GetRandom successful: 0x0:31:64, CRC:0:0
`
trick was the command (with correct flags): {0x02, 0xB2, 0x04}
{ 0x42, 0xB2, 0x04} is not working

My code
` void loop() {
  if (errorFlag) {
    // get random number
    uint8_t getrandom[] = {0x02, 0xB2, 0x04};
    uint8_t *readBuffer;
    ISO15693ErrorCode rc = nfc.issueISO15693Command(getrandom, sizeof(getrandom), &readBuffer);
    if (ISO15693_EC_OK != rc) {
      Serial.println(F("*** GetRandom failed!"));
      return;
    }

    Serial.print(F("GetRandom successful: 0x"));
    Serial.print(readBuffer[0], HEX);
    Serial.print(":");
    Serial.print(readBuffer[1], HEX);
    Serial.print(":");
    Serial.print(readBuffer[2], HEX);
    Serial.print(", CRC:");
    Serial.print(readBuffer[3], HEX);
    Serial.print(":");
    Serial.println(readBuffer[4], HEX);
    return;
  }
`
So the two random bytes are now in readBuffer[1] and readBuffer[2], CRC16 seems always 0.

Next step is to XOR the random number with known password and transfer SET PASSWORD
{ <Flags> 0xB3 0x04 0x04 <XORed Password> }
and getInventory..

I will report..

Dirk

ATrappmann, 2019-10-23:
Hi Dirk,
 
thats great, congratulations! I'm glad that you are now able to 
communicate.
The first 3 errors look strange though. Probably the device needs some 
setup time if it gets into the NFC field!?
You could try to add a little delay() after nfc.begin() or after 
nfc.reset(). Should be something said about the timing in the specs though.
 
Remember, that you ALWAYS have to send a flag byte infront of every 
command. SET PASSWORD with { 0xB3 0x04 0x04 } will not work either. It 
should have a flag byte first: { 0x02 0xB3 0x04 0x04 }!
 
--Andreas.

protected tag (#17)
Datum: 2019-10-22T22:31:46+0200
 
 
 

{ 0xB3 0x04 0x04 }
"https://github.com/ATrappmann/PN5180-Library/issues/17?email_source=notifications\u0026email_token=AEVOSHQKC3XC6FKIH756PITQP5PKZA5CNFSM4JDFHCPKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEB7DAKY#issuecomment-545140779", 
"https://github.com/ATrappmann/PN5180-Library/issues/17?email_source=notifications\u0026email_token=AEVOSHQKC3XC6FKIH756PITQP5PKZA5CNFSM4JDFHCPKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEB7DAKY#issuecomment-545140779", 
"https://github.com" } } ]

tueddy, 2019-10-23:
Hi Andreas,

the first 3 errors wereno errors, the tag was just to far away to communicate. The GetRandom command works absoluty reliable.
I'll try the SetPassword command and give feedback here..
Best 
Dirk

tueddy, 2019-11-05:
Work's fine now, tag is unlocked, preparing a PR..

tueddy, 2019-11-07:
pull request #18

ATrappmann, 2019-11-13:
Thank you for your work! I merged the changes to the library.
--Andreas.

protected tag (#17)
Datum: 2019-11-07T21:15:39+0100
 
 
 

"https://github.com/ATrappmann/PN5180-Library/issues/17?email_source=notifications\u0026email_token=AEVOSHTXNDQD3BHM6XQB2S3QSRZN5A5CNFSM4JDFHCPKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEDNVK3I#issuecomment-551245165", 
"https://github.com/ATrappmann/PN5180-Library/issues/17?email_source=notifications\u0026email_token=AEVOSHTXNDQD3BHM6XQB2S3QSRZN5A5CNFSM4JDFHCPKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEDNVK3I#issuecomment-551245165", 
"https://github.com" } } ]

closed 2019-11-13 by ATrappmann