ATrappmann/PN5180-Library #20: Skipped block data

vladansw, 2019-12-07:
Hello,

first of all thank you very much for writing this library.

Using your example I was able to write 81 82 83 84 to block 0
 
Now I have some troubles with reading entire block data. 
Using your example I get this:
Inventory successful, UID=E0:4:15090D3B05C
----------------------------------
System Info retrieved: blockSize=4, numBlocks=28
----------------------------------
Read block #0: 82 83 84 90  ....

When I use NXP TagInfo app, correct data is received:
[0] . 81 82 83 84

What am I doing wrong?

vladansw, 2019-12-07:
My bad,
blockData[i] = resultPtr[i+1];
fixed the issue

closed 2019-12-07 by vladansw

jsAce7, 2020-01-28:
I had the same trubles. 
Thanks for the tip, it would be nice if the library were fixed.