ATrappmann/PN5180-Library #21: Sleep and when detect card wake up

honestrock, 2019-12-11:
I am trying to run this sensor on sleep mode but I could not Pn5180 will sleep and when detect card wake up ? Did any one try this ? 

> 

bool PN5180::switchModeNormal() {
    
 
    uint8_t      wBufferLength;
    uint16_t     wTempRxDataLen = 0;
    uint16_t     bNumExpBytes;
    uint8_t      bDataBuffer[4];
    uint8_t      bRecBuffer[1];
     
    uint16_t wWakeupCounterInMs=0U;
       /* Build the command frame */
    wBufferLength = 0U;
    bDataBuffer[wBufferLength++] = PHHAL_HW_PN5180_SET_INSTR_SWITCH_MODE;
    bDataBuffer[wBufferLength++] = PHHAL_HW_PN5180_SWITCH_MODE_LPCD;
 
    /* wWakeupCounterInMs LSB */
    bDataBuffer[wBufferLength++] = (uint8_t)(wWakeupCounterInMs & 0xFFU);
    /* wWakeupCounterInMs MSB */
    bDataBuffer[wBufferLength++] = (uint8_t)((wWakeupCounterInMs >> 8U) & 0xFFU);
 
    
    //uint8_t cmd[2] = { PN5180_SWITCH_MODE,mode};
    SPI.beginTransaction(PN5180_SPI_SETTINGS);
    transceiveCommand(bDataBuffer, 4);
    SPI.endTransaction();
    return true;

> }``

tueddy, 2019-12-11:
I have not tried but the low power card detection (LPCD) mode but this feature is very interesting!
Your sample code comes from NXP library?
switchMode() command should be easy to implement in this library but what happens then? Do you want to wakeup the PN5180 with a simple tag or an active target like smartphone? spec. is a bit unclear about..

honestrock, 2019-12-11:
.I want to wake up it by using a mifare simple tag. as i understand i need to 4 parameters SWITCH_MODE, SWITCH_MODE(Mode as lpcd),time min val , time up value.

tueddy, 2019-12-11:
PN5180_SWITCH_MODE command is defined in PN5180.cpp line 34, the command should be easy to implement, see command loadRFConfig() how it works in this library.
But how to setup supported protocols and what happens if card is in field? IRQ status, GPIO? do you have more information? page(s) in spec?

honestrock, 2019-12-11:
normally i dont use nxp library , check it for just to see the uses. That's exactly what I want to do. Sleeping Ardunio and PN5180 card. When the Pn5180 detects a card, wake ardunio and read the card. I will use card as ISO 14443 Type A Mifare 1K 13.56.

tueddy, 2019-12-11:
Same as i want to do. See my fork/branch for ISO-14443 / Mifaire support. Current loop for card detection consumes 50-70 mA..
Maybe work for LPCD on this branch and make a pull request here later? I could provide you the switch-mode command:
https://github.com/tueddy/PN5180-Library/tree/ISO14443

honestrock, 2019-12-11:
thank you for help @tueddy , if i success i will send pull reques to your branch.

ATrappmann, 2019-12-12:
Hi Mert,
 
one small thing I saw after comparing your code snippet to the PN5180 
specification:
 
Field ‘Wake-up Counter Value’ () defines the period between two LPCD 
attempts (=time
PN5180 remains in standby) as has to be in the range from 1 to 2690, 
inclusive. No
instructions must be sent while being in this mode. Termination is 
indicated using an
interrupt.
 
That in mind, your counter is out of range:
 
uint16_t wWakeupCounterInMs=0U;
 
Maybe that will change something.
 
Cheers
--Andreas.
 
﻿

tueddy, 2019-12-13:
Hi, i've implemented the function in my branch like this:
```
/* switch the mode to LPCD (low power card detection)
 * Parameter 'wakeupCounterInMs' must be in the range from 0x0 - 0xA82
 * max. wake-up time is 2960 ms.
 */
bool PN5180::switchToLPCD(uint16_t wakeupCounterInMs) {
  uint8_t cmd[4] = { PN5180_SWITCH_MODE, 0x01, (wakeupCounterInMs & 0xFFU), ((wakeupCounterInMs >> 8U) & 0xFFU) };
  SPI.beginTransaction(PN5180_SPI_SETTINGS);
  bool success = transceiveCommand(cmd, 4);
  SPI.endTransaction();

  return success;
}
```
But i'm unable to make a working demo. The only thing i can see is power consumption is droped to 5-7mA, but no LPCD IRQ is raised. Any idea?

tueddy, 2020-01-05:
Hi, 
i managed to get LPCD-mode working. the current is 0.3mA in sleep mode and ca. 6 mA while wakeup. Only ISO14443 cards can wake up the reader. Unfortunatly the cheap china boards comes with old firmware 3.5, with this firmware i was unable to get this mode working. After upgrading firmware to 4.0 it works. I will provide a demo in my branch. Firmware update was done with this library: 
https://github.com/zGrEaNz/PN5180_Updater

Best regards
Dirk

velinazim, 2020-01-06:
i get such error while uploading. Have you ever had a  error like this? What's your SPI version? @tueddy 

C:\Users\velia\Documents\Arduino\libraries\PN5180_Updater-master\src\PN5180_Firmware.cpp: In member function 'bool PN5180_Firmware::RawTransceive(const uint8_t*, size_t, uint8_t*, size_t)':

C:\Users\velia\Documents\Arduino\libraries\PN5180_Updater-master\src\PN5180_Firmware.cpp:105:42: error: no matching function for call to 'SPIClass::transfer(uint8_t*, const size_t&)'

   SPI.transfer((uint8_t*)TxBuffer, TxSize);

                                          ^

C:\Users\velia\Documents\Arduino\libraries\PN5180_Updater-master\src\PN5180_Firmware.cpp:105:42: note: candidate is:

In file included from C:\Users\velia\Documents\Arduino\libraries\PN5180_Updater-master\src\PN5180_Firmware.h:13:0,

                 from C:\Users\velia\Documents\Arduino\libraries\PN5180_Updater-master\src\PN5180_Firmware.cpp:9:

C:\Users\velia\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.12.31\libraries\SPI\src/SPI.h:286:11: note: uint8 SPIClass::transfer(uint8) const

     uint8 transfer(uint8 data) const;

           ^

C:\Users\velia\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.12.31\libraries\SPI\src/SPI.h:286:11: note:   candidate expects 1 argument, 2 provided

C:\Users\velia\Documents\Arduino\libraries\PN5180_Updater-master\src\PN5180_Firmware.cpp:119:32: error: no matching function for call to 'SPIClass::transfer(uint8_t*&, size_t&)'

   SPI.transfer(RxBuffer, RxSize);

                                ^

C:\Users\velia\Documents\Arduino\libraries\PN5180_Updater-master\src\PN5180_Firmware.cpp:119:32: note: candidate is:

In file included from C:\Users\velia\Documents\Arduino\libraries\PN5180_Updater-master\src\PN5180_Firmware.h:13:0,

                 from C:\Users\velia\Documents\Arduino\libraries\PN5180_Updater-master\src\PN5180_Firmware.cpp:9:

C:\Users\velia\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.12.31\libraries\SPI\src/SPI.h:286:11: note: uint8 SPIClass::transfer(uint8) const

     uint8 transfer(uint8 data) const;

tueddy, 2020-01-06:
Hi velinazim,
i got same compiler errors with the firmware updater. I had success using STM32 bluepill with STM-Link V2 using the "STM32 cores" from arduino board library. SPI libriaries seems incaptible to STM32Duino, see https://github.com/zGrEaNz/PN5180_Updater/issues/1

My LPCD demo is here: https://github.com/tueddy/PN5180-Library/tree/ISO14443/examples/PN5180-LowPowerCardDetection

abidxraihan, 2020-12-18:
Hi,
I managed to get LPCD-mode working. Unfortunately these cheap ebay/aliexpress/amazon boards comes with firmware 3.5. I was unable to get LPCD working. After upgrading firmware to 4.1 it works. Firmware update was done with this library:
[PN5180 Firmware Updater Using ESP32](https://github.com/abidxraihan/PN5180_Updater_ESP32). I had success using ESP32 in an Arduino development environment. Unfortunately I don't have an Arduino lying around to test on. But I don't see why it shouldn't work.

honestrock, 2020-12-19:
@abidxraihan  can you share your lpcd works code ?

abidxraihan, 2020-12-19:
Sure, here is the code [PN5180-LowPowerCardDetection_Firmware 4.1.zip](https://github.com/ATrappmann/PN5180-Library/files/5720004/PN5180-LowPowerCardDetection_Firmware.4.1.zip). The library that was used was Tueddy's [https://github.com/tueddy/PN5180-Library](url).

tueddy, 2020-12-19:
@abidxraihan Thanks for sharing! Your ESP-32 updater should make live easier!

Can you tell what is different in your demo to [mine](https://github.com/tueddy/PN5180-Library/blob/master/examples/PN5180-LowPowerCardDetection/PN5180-LowPowerCardDetection.ino)? I see another treshold but no other important diffs. Can you share some experience using LPCD?
Did you manage to make a demo using ESP-32 in deep sleep mode? This would save power!

Thanks & best regards
tueddy

tueddy, 2020-12-22:
@abidxraihan : Your LPCD demo runs fine, thanks!
Did you manage to get the LPCD demo running with deep sleep? Only then the whole system is running with low power and makes sense!
When i go to deep sleep in ESP32 the system wakes up immediatly even no card is present. Don't know where the wakeup comes from..

tueddy, 2021-01-16:
I have added a simple [demo ](https://github.com/tueddy/PN5180-Library/blob/master/examples/PN5180-LPCD-ESP32-deep-sleep/PN5180-LPCD-ESP32-deep-sleep.ino) showing wake-up an ESP-32 from deep-sleep using LPCD.

The trick is to freeze the pin states before going to deep sleep:
```
      // freeze pin states in deep sleep
      gpio_hold_en(gpio_num_t(PN5180_NSS)); // NSS
      gpio_hold_en(gpio_num_t(PN5180_RST)); // RST
      gpio_deep_sleep_hold_en();
```
Power consumption is ~300uA in deep-sleep mode. This video shows usage in a kids audio player. if metal is on reader, the ESP-32 also wakes up but player software decides to goe to sleep again (false alarm):

https://user-images.githubusercontent.com/11274319/104822749-34fe4980-5845-11eb-92fb-2200bc3177c2.mov

adelgadokeya, 2021-04-26:
Hi, 

I am designing a low power device that uses PN5180. And can not reach less than 200uA on LPCD with 500ms RF on.

Anyone have idea how to have less consumption? The amount of 200uA are from VBAT pins. 

Thanks