ATrappmann/PN5180-Library #23: running on ESP8266

Phreak87, 2020-01-31:
Hi,

yesterday i tried to setup the PN5180 on my ESP8266 development board.
but i can not get it running because of different crashes of the ESP if i try
to access your library.
Your Library contains only definitions for ESP32. 
Is there any Problem with ESP8266 or is it because i do something wrong?
I compiled using the latest version of ESPTool for the latest Arduino IDE.

Thank you

ATrappmann, 2020-01-31:
The library was developed for a plain Arduino. The addition for ESP32 was done by the community.

I never worked with an ESP8266, so that I cannot provide any help.

Sorry.

-Andreas.

Phreak87, 2020-02-04:
Hi,

after a few test i found a way to use it on the ESP8266 Dev-Board:

3.3V <-> 3.3V
GND <-> GND
BUSY <-> D0
RST    <-> D1
NSS   <-> D2
SCK <->   D5
MISO <-> D6
MOSI <-> D7

and use this define Block (if you work in the Arduino IDE).
This is because the Pin numbers are wrong for ESP8266 in the
IDE.

#define PN5180_NSS  4
#define PN5180_BUSY 16
#define PN5180_RST  5

after this it should work.

closed 2020-02-04 by Phreak87