ATrappmann/PN5180-Library #24: Won't go passed PN5180 Hard-Reset

MagicEk, 2020-02-11:
Hi, thank you for sharing your work. 
I'm trying to get my PN5180 to work with the example code.
I've wired as instructed but when I run the code it won't go passed "PN5180 Hard-Reset..." no matter what I tried.  Tried to change to another arduino, diffrent PN5180 and voltage divider, without luck.

P.S I'm a beginner with both codeing and electric components.

Thanks in advance.
Best regards, Erik.

ATrappmann, 2020-02-12:
Hi Erik,
 
did you have a token on the card reader, when the arduino starts up?
 
After the reset, the library only waits for an idle interrupt of the PN5180 
module. Right after that is the next text output written to the serial 
monitor.
 
I think you should start without any token on the card reader!
 
Regards,
--Andreas.

(#24)
 
 
 

"https://github.com/ATrappmann/PN5180-Library/issues/24?email_source=notifications\u0026email_token=AEVOSHU5EGXFRHUGR264X3TRCLI4TA5CNFSM4KTEM3SKYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4IMT2NXQ", 
"url": 
"https://github.com/ATrappmann/PN5180-Library/issues/24?email_source=notifications\u0026email_token=AEVOSHU5EGXFRHUGR264X3TRCLI4TA5CNFSM4KTEM3SKYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4IMT2NXQ", 
"name": "View Issue" }, "description": "View this Issue on GitHub", 
"publisher": { "@type": "Organization", "name": "GitHub", "url": 
"https://github.com" } } ]

TobyBorland, 2020-05-26:
Hello folks,

I had the exact same issue which I solved. I am using an Olimexino-32U4, an Arduino Leonardo clone which allows me to run signals at 3v3.
The SPI hardware lines of the Leonardo are not shared with the edge headers as in the Uno. Instead they are only connected to the 6-pin ISCP header. Once D14-MISO, D15-SCK, D16-MOSI are connected to the relevant PN5180 header pins, everything works. Note BUSY/SS/RESET may be mapped as needed.

Regards all,
Toby