ATrappmann/PN5180-Library #26: Error compiling sketch in Arduino IDE

Slava0096, 2020-03-16:
Hello.
Compiling an example PN5180-Library.ino. Gives an error message. What needs to be done to fix it?

Arduino: 1.8.12 (Linux), Плата:"Arduino Nano, ATmega328P (Old Bootloader)"

/home/slava/Arduino/libraries/PN5180-Library-master/PN5180FeliCa.cpp: In member function 'uint8_t PN5180FeliCa::pol_req(uint8_t*)':
/home/slava/Arduino/libraries/PN5180-Library-master/PN5180FeliCa.cpp:82:26: error: no matching function for call to 'PN5180FeliCa::readData(int, uint8_t*&)'
  if (!readData(20, buffer))
                          ^
In file included from /home/slava/Arduino/libraries/PN5180-Library-master/PN5180FeliCa.h:22:0,
                 from /home/slava/Arduino/libraries/PN5180-Library-master/PN5180FeliCa.cpp:22:
/home/slava/Arduino/libraries/PN5180-Library-master/PN5180.h:102:13: note: candidate: uint8_t* PN5180::readData(int)
   uint8_t * readData(int len);
             ^~~~~~~~
/home/slava/Arduino/libraries/PN5180-Library-master/PN5180.h:102:13: note:   candidate expects 1 argument, 2 provided
exit status 1

ATrappmann, 2020-03-17:
Hi Slava,
 
sorry that was an issue with some code I merged from another user.
I fixed PN5180FeliCa.cpp.
 
The problem should be gone now.
 
Regards
--Andreas.

(#26)
 
 
 
 

"https://github.com/ATrappmann/PN5180-Library/issues/26", "url": 
"https://github.com/ATrappmann/PN5180-Library/issues/26", "name": "View 
Issue" }, "description": "View this Issue on GitHub", "publisher": { 
"@type": "Organization", "name": "GitHub", "url": "https://github.com" } } 
]

Slava0096, 2020-03-18:
Hello.
Now everything is working.
Thank you very much.

closed 2020-03-18 by Slava0096