ATrappmann/PN5180-Library #30: Compiling with platformio

Mopele, 2020-06-13:
Hey, I have downladed the libary via platformio and also as the zip from github. When I upload the file via the arduino IDE everything runs fine, but everytime I use platformio I get those errors. Maybe you know what to do!
Here are the errors. I am using a slightly modified version of the example:
In file included from src\main.cpp:13:0:
C:/Users/Moritz Pfennig/.platformio/lib/PN5180 Library_ID5843/PN5180ISO15693.h:65:9: error: '__FlashStringHelper' does not name a type
   const __FlashStringHelper *strerror(ISO15693ErrorCode errno);
         ^
src\main.cpp: In function 'void loop()':
src\main.cpp:145:24: error: 'class PN5180ISO15693' has no member named 'strerror'
     Serial.println(nfc.strerror(rc));
                        ^
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1`
`

ATrappmann, 2020-06-15:
Hi,
 
sorry I do not know platfromio. FlashStringHelper is specific for the 
ArduinoIDE environment. 
Because your problems are only in the error handling part, comment out the 
lines or exchange them with your own implementation.
--Andreas.

 
 
 
 

"https://github.com/ATrappmann/PN5180-Library/issues/30", "url": 
"https://github.com/ATrappmann/PN5180-Library/issues/30", "name": "View 
Issue" }, "description": "View this Issue on GitHub", "publisher": { 
"@type": "Organization", "name": "GitHub", "url": "https://github.com" } } 
]

Mopele, 2020-06-15:
> 
> Hi, sorry I do not know platfromio. FlashStringHelper is specific for the ArduinoIDE environment. Because your problems are only in the error handling part, comment out the lines or exchange them with your own implementation. --Andreas.
> […](#)

Yes, that is exactly what I did.  I just commented the lines out.
Another question I have concernig this libary is, if it would be possible to get the id as only numbers? Rightnow you get the id as a somehow weird format.

bettse, 2020-09-15:
I have a platformio project that uses this library if you need a reference: https://gitlab.com/bettse/iclassreader/-/tree/PN5180