ATrappmann/PN5180-Library #32: Error to Write into TI RF-HDT-DVBB-N2 Tag

ambrosekwok, 2020-07-06:
Hello Andreas, thanks for your great library. Just simply use the PN5180-Library.ino example with ESP32 board. 
This work great except when I try to write to RF-HDT-DVBB-N2 tag from Texas Instruments. I can read the tag, but cannot write to it.

Would you please advise what could I do to make it work to write to this tag? Many thanks.

Here are the logs from Arduino IDE's Serial Monitor:

> Uploaded: Jul  6 2020 21:52:12
> PN5180 ISO15693 Demo Sketch
> ----------------------------------
> PN5180 Hard-Reset...
> ----------------------------------
> Reading product version...
> Product version=3.5
> ----------------------------------
> Reading firmware version...
> Firmware version=3.5
> ----------------------------------
> Reading EEPROM version...
> EEPROM version=145.0
> ----------------------------------
> Enable RF field...
> ----------------------------------
> Loop #0
> Inventory successful, UID=E0:7:8031E3335419
> ----------------------------------
> System Info retrieved: blockSize=4, numBlocks=64
> ----------------------------------
> Read block #0: 00 00 00 00  ....
> Read block #2: 00 00 00 00  ....
> Read block #3: 00 00 00 00  ....
> ......
> ......
> ......
> Read block #61: 00 00 00 00  ....
> Read block #62: 00 00 00 00  ....
> Read block #63: 00 00 00 00  ....
> ----------------------------------
> Error in writeSingleBlock #0: Option is not supported!
> IRQ-Status 0x4007: [ RQ TX IDLE RX_SOF_DET ]
> ----------------------------------
> 

And below are the scanned info for the RFID tag using NXP TagInfo Android app:

> ** TagInfo scan (version 4.24.6) 2020-07-06 22:43:51 **
> Report Type: External
> 
> -- IC INFO ------------------------------
> 
> # IC manufacturer:
> Texas Instruments
> 
> # IC type:
> Tag-it HF-I Plus (TMS37112)
> 
> -- NDEF ------------------------------
> 
> # No NDEF data storage populated:
> 
> -- EXTRA ------------------------------
> 
> # Memory size:
> 256 bytes
> * 64 blocks, with 4 bytes per block
> 
> # IC detailed information:
> Supported read commands:
> * Single Block Read
> * Multiple Block Read
> * Get System Information
> AFI supported
> DSFID supported
> IC reference value: 0x8B
> 
> -- FULL SCAN ------------------------------
> 
> # Technologies supported:
> ISO/IEC 15693-3 compatible
> ISO/IEC 15693-2 compatible
> 
> # Android technology information:
> Tag description:
> * TAG: Tech [android.nfc.tech.NfcV, android.nfc.tech.NdefFormatable]
> * Maximum transceive length: 253 bytes
> 
> 
> # Detailed protocol information:
> ID: E0:07:80:31:E3:33:54:19
> AFI: 0x00
> DSFID: 0x00
> 
> # Memory content:
> [00] . 00 00 00 00 |....|
> [01] . 00 00 00 00 |....|
> [02] . 00 00 00 00 |....|
> ......
> ......
> ......
> [3D] . 00 00 00 00 |....|
> [3E] . 00 00 00 00 |....|
> [3F] . 00 00 00 00 |....|
> 
>   x:user locked, *:factory locked, .:unlocked
> 
> --------------------------------------

Thank you.

Ambrose

ATrappmann, 2020-07-07:
Hello Ambrose,
 
from the error message "option not supported", I would guess there is 
something wrong in the write command itself.
According to the datasheet, block numbers start with #1 and not with #0 for 
your tag. Which is strange, but it is always worth a try. Change the write 
loop and start with writing blockNo=1.
 
2nd thought: Remove the 'return' command in the error path, so that the 
sketch tries to write all 64 blocks and see what happens.
 
If both fail, try to set the Option bit in the request flags of the 
writeSingleBlock command. The current value is 0x22, change it to 0x62.
 
Regards,
--Andreas.

Tag (#32)
Datum: 2020-07-06T17:25:03+0200
 
 
 
 

"https://github.com/ATrappmann/PN5180-Library/issues/32", "url": 
"https://github.com/ATrappmann/PN5180-Library/issues/32", "name": "View 
Issue" }, "description": "View this Issue on GitHub", "publisher": { 
"@type": "Organization", "name": "GitHub", "url": "https://github.com" } } 
]

ambrosekwok, 2020-07-12:
Hello Andreas,

Thank you. Change the request flags from 0x22 to 0x62 make it work.
There is "No card detected" error returned, but data wrote into the tag successfully.
And the write loop can start with writing blockNo=0.

> ----------------------------------
> Error in writeSingleBlock #0: No card detected!
> IRQ-Status 0x6: [ TX IDLE ]
> *** No card detected!
> ----------------------------------
> Loop #50
> Inventory successful, UID=E0:7:8031E3335419
> ----------------------------------
> System Info retrieved: blockSize=4, numBlocks=64
> ----------------------------------
> Read block #0: 80 81 82 83  ....
> ......

Ambrose

closed 2020-07-14 by ambrosekwok