ATrappmann/PN5180-Library #39: Sending RFID commands - issueISO15693Command is private

yaneevH, 2021-03-08:
Hello,

I am trying to use this code and the PN5180 to configure and then read commands from the MLX90129 Sensor Tag.

As per [page 14 of this pdf](https://e2e.ti.com/cfs-file/__key/communityserver-discussions-components-files/667/90129_5F00_AN_5F00_Sensor_5F00_tag_5F00_001.pdf) I need to send specific commands to configure the sensors before reading first.  Below is a screenshot of the first couple commands:

[image]

I am attempting at issuing just the first line right now as follows: 

```
void mlxConfig(){
  uint8_t cmd[] = {0x43, 0x21, 0x09, 0x71, 0x70};
  uint8_t *readBuffer_0;
  nfc.issueISO15693Command(cmd,  sizeof(cmd), &readBuffer_0);

  nfc.sendData(cmd,sizeof(cmd));
}

```
But am getting the following errors:

```
error: 'ISO15693ErrorCode PN5180ISO15693::issueISO15693Command(uint8_t*, uint8_t, uint8_t**)' is private
   ISO15693ErrorCode issueISO15693Command(uint8_t *cmd, uint8_t cmdLen, uint8_t **resultPtr);
                     ^
MLX_Comm:326: error: within this context
   nfc.issueISO15693Command(cmd,  sizeof(cmd), &readBuffer_0);
```

Is this not the way that that commands should be sent? If I put the data into the _sendData_ function it compiles fine.

ATrappmann, 2021-03-09:
Hello Yaneev,
 
I think both ways - as you tried - are wrong. The configuration you have to 
do upfront is not part of the ISO15693 standard.
Therefor the command issueISO15693Command does not work. Using sendData 
instead was a good guess, but as shown
in the screenshot, I think you have to write to the EEPROM at certain 
addresses instead of sending data.
 
Personaly I have never had any use case to do so. That is why the class 
PN5180 is missing the method writeEEprom and
only has an implementation to read from the EEPROM.
 
If you want to configure your tags, I think you have to implemented this 
missing function.
 
From the PN5180 datasheet the command must be:
 
The call has to be implemented as follows:
 
Feel free to add such a function to the library and let me know, so I can 
integrated it into the code.
 
Best Regards,
--Andreas.

issueISO15693Command is private (#39)
Datum: 2021-03-09T00:19:30+0100
 
 
 
 

<https://e2e.ti.com/cfs-file/__key/communityserver-discussions-components-files/667/90129_5F00_AN_5F00_Sensor_5F00_tag_5F00_001.pdf> 
<https://user-images.githubusercontent.com/6404325/110394361-4f30f880-803a-11eb-8a7a-a4c6ddc28281.png>

"https://github.com/ATrappmann/PN5180-Library/issues/39", "url": 
"https://github.com/ATrappmann/PN5180-Library/issues/39", "name": "View 
Issue" }, "description": "View this Issue on GitHub", "publisher": { 
"@type": "Organization", "name": "GitHub", "url": "https://github.com" } } 
]

yaneevH, 2021-03-09:
I think maybe an image you meant to post didn't upload?

[image]

I can for sure try writing the writeEEprom method. But since I am still not completely understanding the code (and my C++ abilities are limited), would you be able to give a quick outline of how this method might look like and what aspects would go into creating it?

Thank you!

ATrappmann, 2021-03-12:
Hallo Yaneev,
 
sendingData is a different command and does not write to the EEPROM!
 
Try something like this:
 
#define PN5180_WRITE_EEPROM 0x06
 
bool writeEEPROM(uint8_t addr, uint8_t *buffer, int len) {
  uint8_t cmd[2] = { PN5180_WRITE_EEPROM, addr };
 
  SPI.beginTransaction(PN5180_SPI_SETTINGS);
  bool rc = transceiveCommand(cmd, 2, buffer, len);
  SPI.endTransaction();
 
  return rc;
}
 
Regards,
--Andreas.

Datum: 2021-03-09T08:24:55+0100
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
<https://e2e.ti.com/cfs-file/__key/communityserver-discussions-components-files/667/90129_5F00_AN_5F00_Sensor_5F00_tag_5F00_001.pdf> 
<https://user-images.githubusercontent.com/6404325/110394361-4f30f880-803a-11eb-8a7a-a4c6ddc28281.png>

[ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage",