ATrappmann/PN5180-Library #40: PN5180 EEPROM new value dont writed to EEPROM

alansbor, 2021-06-21:
Hello developers

In my example, can't I write a new value to the EEPROM? 
I did not find an example for recording anywhere in the description.
How to write a new value to the EEPROM correctly? 

 //---------------------------------------------------------------------------
  // 0x34 LPCD_REFERENCE_VALUE RW 1
  //#define LPCD_REFERENCE_VALUE  (0x34)
  Serial.println(F("----------------------------------"));
  Serial.println(F("Reading LPCD_REFERENCE_VALUE ..."));
  uint8_t LPCD_REFERENCE_VALUE_v[1];
  nfc.readEEprom(LPCD_REFERENCE_VALUE, LPCD_REFERENCE_VALUE_v, sizeof(LPCD_REFERENCE_VALUE_v));
  Serial.print(F("LPCD_REFERENCE_VALUE= "));

  Serial.print(LPCD_REFERENCE_VALUE_v[0], HEX);
  Serial.println(" ");
    
  // OLD Value LPCD_REFERENCE_VALUE= 8  dec 8   New value LPCD_REFERENCE_VALUE= AC dec 172
  uint8_t LPCD_REFERENCE_VALUE_vv[1];
  LPCD_REFERENCE_VALUE_vv[0] = 172;
  nfc.writeEEPROM(LPCD_REFERENCE_VALUE, LPCD_REFERENCE_VALUE_vv, sizeof(LPCD_REFERENCE_VALUE_vv));
  

  Serial.println(F("----------------    CHECK NEW VALUE   ------------------"));
  Serial.println(F("Reading LPCD_REFERENCE_VALUE ..."));
  nfc.readEEprom(LPCD_REFERENCE_VALUE, LPCD_REFERENCE_VALUE_v, sizeof(LPCD_REFERENCE_VALUE_v));
  Serial.print(F("LPCD_REFERENCE_VALUE= "));

  Serial.print(LPCD_REFERENCE_VALUE_v[0], HEX);
  Serial.println(" ");
    
  exit(- 1);

  //---------------------------------------------------------------------------

Alexey.

tueddy, 2021-06-21:
If you want to do Low power card detection: have you checked my [fork ](https://github.com/tueddy/PN5180-Library/tree/master/examples/PN5180-LowPowerCardDetection) ?
Best Dirk

alansbor, 2021-06-22:
Hello Dirk !

Thanks a lot for your fork. 

Your fork maintains a correct with writeEEPROM, 
Andreas Trappmann doesn’t write it correctly, so it doesn’t work. 

Alexey.