ATrappmann/PN5180-Library #42: Issue in library for locking SLIX2

Nettermann90, 2021-08-18:
There is an Issue in the Enable Privacy on SLIX2 Tags.

In file:
PN5280ISO15693.cpp

function:
ISO15693ErrorCode PN5180ISO15693::enablePrivacy(uint8_t *password, uint8_t *random)

you need to replace the line:
uint8_t setPrivacy[] = {0x02, 0xBA, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};

with:
uint8_t setPrivacy[] = {0x02, 0xBA, 0x04, 0x00, 0x00, 0x00, 0x00};

there are two bytes to much, with the origianl code it will not work. 

Best Regards, 
Felix

tueddy, 2021-08-18:
You're right! It's already fixed in this fork:
https://github.com/tueddy/PN5180-Library/blob/master/PN5180ISO15693.cpp

[image]

ATrappmann, 2021-08-18:
Thanks for the information. I will fix it asap.
--Andreas.

(#42)
 
 
 
 

[grafik]
<https://user-images.githubusercontent.com/11274319/129878070-c4e1e3e2-682f-4a2c-ac78-1280b6d2b3c9.png>
or Android
[ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage", 
"https://github.com/ATrappmann/PN5180-Library/issues/42#issuecomment-900983697", 
"url": 
"https://github.com/ATrappmann/PN5180-Library/issues/42#issuecomment-900983697", 
"name": "View Issue" }, "description": "View this Issue on GitHub", 
"https://github.com" } } ]

ATrappmann, 2021-08-19:
Fixed in Version 1.8

closed 2021-08-19 by ATrappmann