ATrappmann/PN5180-Library #48: card not detected

lenkapolaskova, 2021-12-01:
May I ask a question? I realized that in case a card is not detected you reset module and repeat whole transaction. Why? I tried to wait for the card in "wait for data" state but card can't be detected. It seems like some kind of a workaround. Is there an issue with the module?

ATrappmann, 2021-12-09:
You are right. The reset is done unintentionally, because the state EC_NO_CARD is part of the return codes. This could be filtered out, so that no reset is done. On the other hand, if a real error occurces, the reset seems to be a good idea to get everything back to a defined state.
So always resetting the device is not part of a workaround of some other issue.

ATrappmann, 2021-12-10:
What do you mean be sending a "reqA command"? Which library function are you using?

tueddy, 2021-12-10:
REQA is an ISO-14443 comannd and implemented here:
`
uint8_t PN5180ISO14443::activateTypeA`

REQA = request 14443 TypeA card
WUPA = wakeup  14443 TypeA card

lenkapolaskova, 2021-12-13:
Hi, I follow AN12650 - Using PN5180 without library. in this guide there are steps to send REQA command. I am stuck on point waitForCardResponse because if the card is not present immediatelly it is not possible to read it then.
Have you tested reading a card without reseting the device?

tueddy, 2021-12-13:
In my fork there are some demos for ISO-14443 card polling: https://github.com/tueddy/PN5180-Library/blob/master/examples/PN5180-ISO14443/PN5180-ISO14443.ino
and a sample for polling both ISO-15693 / 14443 cards: https://github.com/tueddy/PN5180-Library/blob/master/examples/PN5180-ReadUID/PN5180-ReadUID.ino

Maybe a starting point to get rid off resetting the PN5180 and optimize polling times?

lenkapolaskova, 2021-12-16:
In the examples there is nfc.reset() and nfc.setupRF() commands in the loop. I wonder to know why? The reset and setup should be only in the setup part am I right?

tueddy, 2021-12-20:
If reader receives corrupted response due to card gets out of field the PN5180 can get unresponsible.
14443 cards that are in field but do not response to REQA have to be resetted by disable/reactivate the RF field.
Therefore loop makes a complete reset + field reactivation to be on the safe side. This is slow and maybe not everytime needed. 

@lenkapolaskova Feel free to optimize the loop to get a better/faster card recognition!